DuPPadotnet/I2CEncoderV2.1 #4: Dead time after resetting

therealprof, 2019-01-06:
I'm writing a driver in Rust for the I2CEncoderV2 and noticed that my initialisation sequence does not work because as a good citizen I would like to do a RESET followed by proper initialisation but when I set the reset bit the following I2C interaction automatically receives a NACK.

Fattoresaimon, 2019-01-06:
Hello,
I just did some tests.
I have found that the board requires at least 400uS for RESET.
Try to put a 400uS after issuing the RESET command.
Sorry i never had measured it before, i will add in the datasheet. 
Thank you.

therealprof, 2019-01-06:
Hm, no idea how to achieve that in a generic driver. Can I keep polling until it comes back?

Fattoresaimon, 2019-01-06:
You can try, but i'm not sure that is the way.
Remember that the I2C Encoder V2 is based on a MCU and not on FPGA or CPLD.
So it's a bit slower on some operations.

therealprof, 2019-01-06:
Yeah, but I2C is not super fast anyway. I'd totally expect that any MCU can handle the timing by setting I2C up early and stretching the clock until it is ready to handle additional data.

Fattoresaimon, 2019-01-06:
Clock stretch is not enabled, check the other issue https://github.com/Fattoresaimon/I2CEncoderV2/issues/3

Fattoresaimon, 2019-01-06:
The RESET command is a full reset of the MCU.
So after that the MCU will restart from the beginning: initialize the peripherals, the variables, ecc..

therealprof, 2019-01-06:
Sure. But maybe if you want to have a proper I2C slave you should try to prioritise the functionality required for an I2C slave first?