DuPPadotnet/I2CEncoderV2.1 #5: Cannot use RGB values with RGB encoders

therealprof, 2019-01-06:
First I'm enabling the RGB LED by setting the ETYPE bit and then I'm calling a function which tries to do a write on `RLED`, then `GLED` then `BLED` in succession but depending on the combination of colours and the previous value this yields completely random results; single colours mostly work, combinations of RGB deliver completely random results.

Reading back the `RLED`, `GLED` and `BLED` registers shows the correct colours I've set before. Fading is disabled.

Fattoresaimon, 2019-01-06:
Hello,
Can you please post the code with a combination of colors?
Can you please tell me which board you are using?
I've never encountered this problem, so it's look strange.

Fattoresaimon, 2019-01-06:
I have tried this sequence:

Encoder.begin(INT_DATA | WRAP_ENABLE | DIRE_LEFT | IPUP_ENABLE | RMOD_X1 | RGB_ENCODER);
Encoder.writeLEDR(0xff);
Encoder.writeLEDG(0xff);
Encoder.writeLEDB(0xff); 

And i get the LED white, that is correct.
I'm thinking that you are sending the commands to much fast and this causes some misbehavior.
I'm using an ESP8266 and between the commands there are 20uS delay,

therealprof, 2019-01-06:
Uhm, too fast? The weird thing is that the values read back perfectly fine so they're registered by the MCU, just not properly applied.

Fattoresaimon, 2019-01-06:
Try to add a delay between the commands and tell me if it's solve.
And also try to send the commands with the I2C auto increment like the following:

1. Write I2C address of your board
2. Write the  REG_RLED address ->   0x18
3. Write the RLED data -> 0xFF
4. Write the GLED data -> 0xFF
5. Write the BLED data -> 0xFF

therealprof, 2019-01-06:
> Try to add a delay between the commands and tell me if it's solve.

I can't. This is a generic driver, could be executed on anything from a small MCU over a RaspPi over to a Windows server connected via a FTDI chip. I can only use defined features of the I2C bus which is write and write followed by a read.

> And also try to send the commands with the I2C auto increment like the following:

That makes it even worse. Reminds me, this should also be mentioned in the documentation; I wasn't sure if the registers are auto-incrementing since it worked even worse than before when I tried.

But I figured out a way to make it work: Reduce the I2C clock to 10kHz and write some dummy value (e.g. 0, 0, 0) to the RGB LED registers first...

Something is really off here...

Fattoresaimon, 2019-01-07:
Hi,
I have just tried with a NUCLEO-F446RE, but since i don't know how to program in Rust, i have made a simple example in C.
I have used the STM32CubeMX to generate the peripheral code and i have made a simple program in C using eclipse and the System Workbench plugin.
I have set the I2C at 400kHz in fast mode.
Here is the code:
```
  uint8_t buffer[5];
  buffer[0]=0;
  buffer[1]=0b00111111;
  HAL_I2C_Master_Transmit_IT(&hi2c1,0xC2,buffer,2);
  HAL_Delay(2); // Necessary in order to complete the previews writing
  buffer[0]=0x18;
  buffer[1]=0xFF;
  buffer[2]=0xFF;
  buffer[3]=0xFF;
  HAL_I2C_Master_Transmit_IT(&hi2c1,0xC2,buffer,4);
```

In my test, the RGB LED correctly turns on with white color, i did't have any issue.
Here  you can see the I2C transaction, there isn't any delay between the packet, and the RGB LED works correctly.

[image]

I don't know what was happening to you, did you tried with a different board?

Fattoresaimon, 2019-01-23:
Any update?
Did you have some code that i can try?