DuPPadotnet/I2CEncoderV2.1 #6: RMOD not documented

therealprof, 2019-01-06:
There's a RMOD bit in the general configuration, but what does `Reading Mode` `X2 mode` and `X1 mode` actually mean?

Fattoresaimon, 2019-01-06:
Hello, 
They are two types of encoding in the encoders.
Just google a bit: https://www.motioncontroltips.com/faq-what-do-x1-x2-and-x4-position-encoding-mean-for-incremental-encoders/

therealprof, 2019-01-06:
@Fattoresaimon So I guess the documentation should mention then that X2 mode doubles the resolution?

Fattoresaimon, 2019-01-06:
Yes!
I made this in order to be compatible with different type of encoders.

therealprof, 2019-01-06:
That is great but useless without the knowledge what it is and which setting is compatible with which encoder, (i.e. what is correct setting for the RGB led encoders and non-RGB encoders?)