DuPPadotnet/I2CEncoderV2.1 #8: GCONF.DIRE bit doesn't to work.

deadsy, 2019-03-02:
I'm using the kickstarter RGB encoder.

Here are my initial config values:

  {REI2C_GCONF, REI2C_GCONF_ETYPE},
  {REI2C_CVAL, 0}, // Counter Value
  {REI2C_CMAX, 32}, // Counter Max value
  {REI2C_CMIN, -32}, // Counter Min value
  {REI2C_ISTEP, 1}, // Increment step value

With GCONF = 0x20 (DIRE=0) turning it clockwise decrements the cval.
With GCONF = 0x24 (DIRE=1) turning it clockwise *still* decrements the cval.

I would expect that:
DIRE=0 a clockwise turn would increment cval.
DIRE=1 a clockwise turn would decrement cval.

Fattoresaimon, 2019-03-02:
Hello,

The DIRE functionality works, just re-tested to make sure.
I have tested with Arduino and with the following configuration:

Encoder.begin(INT_DATA | WRAP_DISABLE | DIRE_RIGHT | IPUP_ENABLE | RMOD_X1 | RGB_ENCODER);
Rotate right CVAL _increment_

Encoder.begin(INT_DATA | WRAP_DISABLE | DIRE_LEFT | IPUP_ENABLE | RMOD_X1 | RGB_ENCODER);
Rotate right CVAL _decrement_

deadsy, 2019-03-03:
Hmm... Started working.
No code changes, just a power cycle.
Don't you hate it when that happens....

closed 2019-03-03 by deadsy