DuPPadotnet/I2CEncoderV2.1 #10: [bug(?)] Triple pressing acts as a double push instantly followed by a single press...

IAmOrion, 2019-05-09:
Triple pressing acts as a double push instantly followed by a single press.  Realistically this should still fall under control of the debouncing methods, given that's it's highly unlikely someone wants to action a double press and then a single press like that.  Ideally, it should not register the 3rd press... so in other words, it should action the double press method and ignore the 3rd press (within the debounce period)

cconde, 2019-07-28:
Might be a "multiple press" event and a register to read the number of presses.
That would fit all situations, the proposed one by IAmOrion and perhaps somebody coding an Easter egg after ten presses :)

Fattoresaimon, 2019-07-28:
Hello,
In the version V2.1 i have fixed this issue.
After detecting a double push there is window of the same amount of the DPPERIOD where the button is not detected.

IAmOrion, 2019-07-28:
> Hello,
> In the version V2.1 i have fixed this issue.
> After detecting a double push there is window of the same amount of the DPPERIOD where the button is not detected.

Do I have to upload a new firmware to the I2C encoder somehow? Or do I only need to update the library? I have the v2.0 board I believe

Fattoresaimon, 2019-07-28:
You have to update the FW on the board. The FW is compatible.
You need a PIC programmer, like the PICKit 3.
In the datasheet it's written how to connect: https://github.com/Fattoresaimon/I2CEncoderV2.1/blob/master/EncoderI2CV2.1_v1.0.pdf