DuPPadotnet/I2CEncoderV2.1 #15: Board as an KiCad symbol

Renha, 2020-06-03:
Hello! I'm designing some board in KiCad and going to use i2c Encoder there. It would be great to have it as a symbol I could insert to my design to get correct pin placement, names, etc., and footprint, all of that for inserting the i2c Encoder board into mine.

Even better is to have different connection variants in same library - for soldering, for inserting, with GP ports also being soldered, or without, in single KiCad library.

Thank you!

Soundpauli, 2024-12-28:
Hey @Renha or @Fattoresaimon i'm looking for a footprint as well - did someone already made one?

Renha, 2024-12-28:
Not me, I've ended up integrating the encoder more deeply into the project and handle it with the main MCU