DuPPadotnet/I2CEncoderV2.1 #20: Encoder not responding after removing JSD connector

nicolasvair, 2024-04-24:
Hi, 

I bought 4 encoder for my project. They came with JSD XH two side so I could plug everything and test my idea.
[image]

Everything was working fine, the lib is great ! So I removed the JSD connector because I needed something thinner for my design (it's a wearable). 

Now one of the encoder doesnt work. I changed the address without luck, tried connectors from the 4 sides and still nothing. 

With the others 3, it seems I lost the INTERRUPT PIN on one side on 2 of them. One is as new.

Any idea how I could check what has been broken and how to solve it ? 

From the outside it seems perfect, no sign of damage, and I don't have any undesired continuity between the pins (like between SDA and SCL, between + and -).

Thank you !

Fattoresaimon, 2024-04-24:
Hello,

Maybe some trace get damage during the removing?

Try to check if there is continuity among the connector of the 4 side.
I mean check if there is continuity among all the SCL, SDA, INT and power pin of the 4 side.
and also double check the short.
Also check if there are the pull-up resistors on the I2C bus.

Thank you,
Simone

nicolasvair, 2024-04-24:
Hi 

thank you for your quick reply !

I have continuity on 4 sides except for the INTERRUPT on one side. 

So I soldered on one of the other side (it is the last in chain because the other 3 are working). 

I tried with and without pull-up but it didn’t help. 

Do you think it could be because one INTERRUPT IS LOST ? 

Is it possible to use it without interrupt ? 
and could I tweak sth using the available GPIO ? 

Thank you

nicolasvair, 2024-04-24:
Ok I finally found it. SDL was broken. I connected a wire straight to micro controller and got it back.. 
Not sure for how long tho

[image]

My first precision soldering 🥹
With iphone as binocular glass and giant soldering iron 😂

Fattoresaimon, 2024-04-24:
Hello,

Oh nice job! ;)

Thank you,
Simone

closed 2024-04-25 by nicolasvair