DuPPadotnet/I2CEncoderV2.1 #21: Store multiple values in registers ?

nicolasvair, 2024-04-27:
Hi, 

I have a question on how the encoder works. I'm doing a bluetooth midi controller and plan to have several pages of controls. 

I read in the docs that the encoder have 4 adress bytes. Does it means that I can store 4 values ? Or are you using the 4 of them for one value ?

Fattoresaimon, 2024-04-27:
Hello,
I think you are referring to the CVAL, CMAX, CMIN and ISTEP register.
They are 32bit register and they store 1 value. They use 4 register because a 32bit number is composed of 4 byte.

Thank you!

nicolasvair, 2024-04-27:
Allright, that what I thought.

I have then another related question. What would be the benefit of storing the value in the register rather than in an arduino variable ? 

Thank you

Fattoresaimon, 2024-04-27:
The I2C encoder keep counting when you rotate the encoder even if you don't read trough the I2C command. This can happens when your arduino is busy with other tasks. 
So in this way you will not miss the steps

nicolasvair, 2024-04-28:
Oh ok. That makes sense. 

Thank you ! Those controllers and led rings are really great and well made.